zhouhan0126/WIFIMANAGER-ESP32 #1: Connection problem only on last esp32 Core

Testato, 2017-11-08:
This library seems not compatible with the last esp32 arduino core Commit
https://github.com/espressif/arduino-esp32

The password is saved but the Connection Result return always 1 (Failed to connect)
```
*WM: WiFi save
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 1
*WM: Failed to connect.
```

If you reset manually the board the connection will established, so the password is correctly stored.
I tried with oldArdiono core version and all return to normal so should be a problem whit some modification on new commit (is there also some IDF sdk update)

eirikso, 2018-01-03:
I fixed this by adding the following lines in WiFiManager.cpp:

Add this at line 216: 
ESP.restart();
delay(1000);

[image]

Brunez3BD, 2018-05-08:
I had the same problem. 
@eirikso can you please push this changes to the master? So in the future we don't have to manually fix this problem..
(I'm new on github and don't know well how to push changes to other's repositories)

domingosl, 2018-05-09:
This does not fix the issue, it is a workaround. I'm doing the same thing in my application, but I will like to understand why sometimes you need to reset the ESP32 even though the SSID and Password were correctly stored.

sajingeo, 2018-05-16:
I have the same problem

Testato, 2018-05-17:
The original WifiManager now support the ESP32, try it

anwarbashir, 2018-06-13:
@Testato Do you mean this one https://github.com/tzapu/WiFiManager?

anwarbashir, 2018-06-13:
The following example code has stopped working for me, It tried the above workaround but I am unable to obtain a WiFi connection, can anyone advise?

//Coded for ESP32 only

#include <WiFi.h>
#include <DNSServer.h>
#include <WebServer.h>
#include <WiFiManager.h>         //https://github.com/tzapu/WiFiManager

void setup() {
    // put your setup code here, to run once:
    Serial.begin(9600);

    //WiFiManager
    //Local intialization. Once its business is done, there is no need to keep it around
    WiFiManager wifiManager;
    //reset saved settings
    //wifiManager.resetSettings();
    wifiManager.autoConnect("AutoConnectAP");
    Serial.println("connected...yeey :)");
}

void loop() {
    // put your main code here, to run repeatedly:
    
}

*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result:
*WM: 1
*WM: SET AP STA
*WM:
*WM: Configuring access point...
*WM: AutoConnectAP
*WM: AP IP address:
*WM: 192.168.4.1
*WM: HTTP server started
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Scan done
*WM: DUP AP: Virgin Media
*WM: DUP AP: Virgin Media
*WM: VM575052-2G
*WM: -38
*WM: Virgin Media
*WM: -38
*WM: BTHomeSpot-RS2
*WM: -87
*WM: VM2135198
*WM: -92
*WM: VM281786-2G
*WM: -93
*WM: Sent config page
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: WiFi save
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Connection result:
*WM: 1
*WM: Failed to connect.
09�␄�␘L0�(JʄN)��9��*WM:
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result:
*WM: 1
*WM: SET AP STA
*WM:
*WM: Configuring access point...
*WM: AutoConnectAP
*WM: AP IP address:
*WM: 192.168.4.1
*WM: HTTP server started

Testato, 2018-06-13:
Yes, the development branch
https://github.com/tzapu/WiFiManager/tree/development

anwarbashir, 2018-06-14:
@Testato, Yeey I am able to connect again. Thank you so much, whilst I still have some issues to resolve at least I have been able to obtain a connection using the dev code brank on my ESP32. I have no idea why the zhouhan library stopped working. I am using PlatformIO and I did not change anything so far as I am aware. Anyway it is great to have a connection again. The dev library is not disconnecting properly but I will investigate further to see if I can find the reason. In the meantime thanks again :)