zhouhan0126/WIFIMANAGER-ESP32 #2: Wifimanager don't open dialog window automaticly when connect to ESP-32 AP

Radeckf, 2017-12-12:
Normaly a popup window comes when PC or smartphone connect to WIFIMANAGER ESP-32 in access-point mode. At my devices this feature not work. If you connect on the IP-address with a webbrowser you are able to configure WIFI, but not automaticly. After analyse packets between my Macbook and ESP32 I found there is no valid DNS-Server entry in DHCP packet. The value is alway "0.0.0.0". After hours of work I found a solution to set the entry manualy. In WiFiAP.cpp at the end around line 140 in function WiFiAPClass::softAP paste the following code:

tcpip_adapter_dns_info_t dns_info;
dns_info.ip.u_addr.ip4.addr = IPAddress(192,168,4,1);
tcpip_adapter_set_dns_info(TCPIP_ADAPTER_IF_AP, TCPIP_ADAPTER_DNS_MAIN, &dns_info); 

and if you set your IP-address manualy in WiFiAPClass::softAPConfig  paste around line 180:

tcpip_adapter_dns_info_t dns_info;
dns_info.ip.u_addr.ip4.addr = info.ip.addr;		
tcpip_adapter_set_dns_info(TCPIP_ADAPTER_IF_AP, TCPIP_ADAPTER_DNS_MAIN, &dns_info);

So the DNS-setting in DHCP-protocol is correct and the configuration window pops up when connecting to WiFIMANAGER

eirikso, 2018-01-02:
Thank you so much. I had the same problem and found this comment. In my version of WiFiAP.cpp I had to paste your lines of code from line 95. It ended up looking like this:

....
bool WiFiAPClass::softAP(const char* ssid, const char* passphrase, int channel, int ssid_hidden, int max_connection)
{
  tcpip_adapter_dns_info_t dns_info;
  dns_info.ip.u_addr.ip4.addr = IPAddress(192,168,4,1);
  tcpip_adapter_set_dns_info(TCPIP_ADAPTER_IF_AP, TCPIP_ADAPTER_DNS_MAIN, &dns_info);

    if(!WiFi.enableAP(true)) {
        // enable AP failed
        return false;
    }
....

And now the portal pops up automatically. Again, thank you.

Loufylouf, 2018-01-18:
@Radeckf have you found any use to the type tcpip_adapter_dns_param_t, that contains tcpip_adapter_dns_type_t and tcpip_adapter_dns_type_t ?

Loufylouf, 2018-01-18:
And instead of hard coding the IP address, you could use the method taken from WiFi.softAPIP : 
`    tcpip_adapter_ip_info_t ip;
    tcpip_adapter_get_ip_info(TCPIP_ADAPTER_IF_AP, &ip);
    return IPAddress(ip.ip.addr);`

Loufylouf, 2018-01-19:
If you haven't noticed, @me-no-dev integrated the DNS server library in the Arduino-esp32 repository (https://github.com/espressif/arduino-esp32/commit/615c9279f79f9f721eedc3d48b3506b3c3054492) and testing with this version, the DNS problems disappears.

Testato, 2018-01-28:
The problem is solved only on Osx,
 Android, Linux, Windows continue to NOT open dialog window automaticly.

anwarbashir, 2018-06-13:
@Radeckf Please could you advise me, there is no file called WiFiAP.cpp in this repository (as far as I can see), can you confirm where this file can be found? Thanks

Radeckf, 2018-06-14:
This file is part of the ESP-32 WiFi library, not of the Wifimanager. You will find this file in the core library.
Regards,
Radeckf

anwarbashir, 2018-06-17:
@Radeckf, Hi Thanks, I found the file here https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiAP.cpp, can you confirm is this correct? Thanks :)

Radeckf, 2018-06-17:
Hi, yes that’s correct
Regards
radeckf

anwarbashir, 2018-06-18:
@Radeckf. Thanks for confirming the location of the library. At present I am using Platform.IO and I have WiFIManager working (subject to further testing) but without the use of this library. I am not sure how to implement this library into Platform.IO and this prevents me from implementing your code, at the moment.
Presently, on my andriod phone the browser dialog does not appear after the ip connects, so I will try and implement your code to resolve this. Any guidance would be gratefully received.

davidjonz2005, 2018-06-28:
i tested in Windows and Android and not worked 

[image]