zhouhan0126/WIFIMANAGER-ESP32 #10: how to prevent esp from going to AP mode after save config if it can not connect to SSID

davidjonz2005, 2018-06-28:
i have some application that i want to configure an ESP with my WIFI then i want it always be in client mode even if it does not see any successful AP. Actually i put a button that will make my ESP go to AP mode (never comeback to AP mode automatically)and reset the module. so what can i do to achieve this?

karan6190, 2018-06-29:
So if you want to send to send you esp into StandAlone mode after saving config then you have forcefully change the WiFi.mode while its checking esp status.

[image]

if (connectWifi(_ssid, _pass) != WL_CONNECTED) {
        DEBUG_WM(F("Failed to connect."));
        WiFi.mode(WIFI_STA);
      }

but just make sure, it will never be in a AP mode again.

davidjonz2005, 2018-08-07:
long time passed 
i ask my question again

where do i have to look in order to find out there is a config already set?
and how can I prevent esp from going to AP mode after checking that a config has already been set?