zhouhan0126/WIFIMANAGER-ESP32 #11: I have a compilation issue, with this library

GhassanYusuf, 2018-06-29:
Dear Friends

good day to you, i have a problem, when trying to compile the code from example files. plz help

[image]

GhassanYusuf, 2018-06-29:
does both header files have the same functions, because it get confused with esp32 and the arduino native WiFi library. so in future if i am using a different board, maybe it will not work.

karan6190, 2018-06-29:
It's not specific to Esp32 and both the header file may have different functionality but names are same, but in your case, you just want Wifi.h for turning its wifi module on, so it's good to take it from espressiff\esp32.

And the above error is because you don't have WebServer lib.
just download https://github.com/bbx10/WebServer_tng and extract in your lib folder

GhassanYusuf, 2018-06-29:
Wow, thanks @karan6190 
it works now, wohoo