zhouhan0126/WIFIMANAGER-ESP32 #14: Cant set hostname

Viniciusvalenca, 2018-08-19:
Hello, thanks for this great work!! 
On my project i tried to set the hostname, but it wont work, on others issues opened about this same subject a user posted that he solved the problem by adding the setHostName after wifi.begin, but when we use the Wifimanager, we dont have this "wifi.begin()", of course this is behind the WifiManager, but maybe need to create a sethostname on this lib? i dont know....

Here is the link about what i said above:

https://github.com/espressif/arduino-esp32/issues/806

luisgcu, 2019-02-09:
I have the same problem..

MHz000, 2019-03-08:
using   
          wifiMulti.addAP(ssid, pass);   
I run into the same problem.
Any hint?

Viniciusvalenca, 2019-03-08:
Nothing....the creator of the plugin did not answer =/ My solution was, don't use this plugin because this dont work right

miguelcsouza, 2020-10-06:
I use the wifimanager library and define my hostname with the command `WiFi.hostname("Test");`inside `void setup()` before even calling any WifiManager function