zhouhan0126/WIFIMANAGER-ESP32 #17: [HTTP HEAD] problem

Cyb3rn0id, 2018-09-08:
Give me this error:

In file included from C:\Users\Gianni\Documents\Arduino\libraries\WiFiManager-ESP32\examples\AutoConnect\AutoConnect.ino:14:0:

C:\Users\Gianni\Documents\Arduino\libraries\WiFiManager-ESP32/WiFiManager.h:36:22: error: 'const char HTTP_HEAD []' redeclared as different kind of symbol

const char HTTP_HEAD[] PROGMEM            = "<!DOCTYPE html><html lang=\"en\"><head><meta name=\"viewport\" content=\"width=device-width, initial-scale=1, user-scalable=no\"/><title>{v}</title>";

                      ^

In file included from C:\Users\Gianni\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\WebServer\src/WebServer.h:30:0,

                 from C:\Users\Gianni\Documents\Arduino\libraries\WiFiManager-ESP32\examples\AutoConnect\AutoConnect.ino:12:

C:\Users\Gianni\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\WebServer\src/HTTP_Method.h:10:3: note: previous declaration 'HTTPMethod HTTP_HEAD'

   HTTP_HEAD    = 0b00100000,

   ^

Garito, 2018-09-13:
Hi!
Same problem here
Anyone taking care of this?
Thanks

Garito, 2018-09-15:
There is an async version of this manager without the issues here (I'm already using it)
https://github.com/alanswx/ESPAsyncWiFiManager

Juliothurler, 2018-09-15:
i will check with your suggestion, thanks a lot , and I will send you how
is this working on, again thank you very much.

CodingCaio, 2018-09-16:
I'm having the same issue here.

@Garito, I just tried your suggestion but the author of WiFiAsyncManager says to add "<WiFiManager.h>", which is this library. I don't get it.

Garito, 2018-09-17:
The guy copied the project and make the modifications but not updated the readme and so
Check the examples
If implemented even the websocket server using them

bobermen1, 2018-10-03:
I was having this issue with the examples as well.
To fix it I replaced
`
#if defined(ESP8266)
#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino
#else
#include <WiFi.h>          //https://github.com/esp8266/Arduino
#endif

//needed for library
#include <DNSServer.h>
#if defined(ESP8266)
#include <ESP8266WebServer.h>
#else
#include <WebServer.h>
#endif
#include <WiFiManager.h>         //https://github.com/tzapu/WiFiManager
`

With
`#include <WiFi.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>`

CodingCaio, 2018-10-08:
@bobermen1, are you using ESP32 or ESP8266?

I attempted replacing all HTTP_HEAD with HTTP_HEADER and it compiles and uploads it but I couldn't get the library to work. I tried this with the AutoConnect example that comes with the library. This sucks.

kotran88, 2018-10-09:
I tried all the things above, but didn't work any.

CodingCaio, 2018-10-09:
Yeh @kotran88, same here. If you have selected the board ESP32, the compiler won't fetch the packages for ESP8266 and vice-versa. Different libraries for different cores. There seems to be a conflict with the variable name HTTP_HEAD and WebServer library.

CodingCaio, 2018-10-10:
To be a little more specific (for the people from the future with similar problem to find), this was "Brownout detector was triggered" flag thrown in the serial monitor. So if you have this problem, try a better cable. It seems to be a common problem when using the WiFi module of the ESP32. If you don't use the WiFi module, things seem peachy. 

Long story long, initially I thought it was a grounding problem with my USB port on my computer, then I thought maybe it was the cable so I went out and bought a new one... which threw me off because the problem wasn't solved and it was a new cable... Also, when I powered the board with a battery bank, everything worked fine except the WiFi. At the end, after reading a bunch of forums and brainstorming I went out and got a better cable, problem solved.

lacneto, 2018-11-04:
I was having the same HTTP_HEAD problem and the solution was simpler than it looked. I just edited the WiFiManager.h files and WifiManager.cpp replacing (HTTP_HEAD) with (HTTP_HEADER) and there was no more compilation problem.

mineirinRV, 2018-12-03:
I had the same problem,
so I located the file "HTTP_Method.h"
in the "C: \ Arduino \ hardware \ espressif \ esp32 \ libraries \ WebServer \ src" folder, modified
 line 10   HTTP_HEAD = 0b00100000,  --> HTTP_HEAD1 = 0b00100000,
and everything worked correctly.
MineirinRV

shootdaj, 2019-01-10:
A better solution imo is to rename the constant in WifiManager.h:

HTML_HEAD -> HTTP_HEAD_HTML

and then rename all the references in WifiManager.cpp:

HTML_HEAD -> HTTP_HEAD_HTML

I verified that this works. This way you are not breaking WebServer if it references that variable anywhere.

Brunez3BD, 2019-02-18:
There is a pull request pending for this problem. I've created a fork and solved this issue, you can find it here [Brunez3BD/WifiManager](https://github.com/Brunez3BD/WIFIMANAGER-ESP32)

FRG7854, 2019-04-05:
Brunez3BD,
Thank you very much.
I found my WiFiManager files, opened it with Wordpad, replaced with your program and everything is fine.
Thank you again.

dalwinder1578singh, 2019-09-02:
i am facing problem with esp32 , it was running very good uploading debugging, all of sudden warning came no more support esp board, i have tried all out, re installing ide, reinstalling ,packages,now every thing looks ok board is read on screen , up load ok data verified no error message, but application does not start, can any one give cue what is going one i have 4 esp board all of them same result. thanks in advance.

jotadepicas, 2020-04-05:
@shootdaj thanks for the workaround, it worked for me. However, could there be a typo in your message? I think it's not:

**HTML**_HEAD -> HTTP_HEAD_HTML

but actually:

**HTTP**_HEAD -> HTTP_HEAD_HTML.

israeloriente, 2020-05-21:
> I was having this issue with the examples as well.
> To fix it I replaced
>
> ```
> #if defined(ESP8266)
> #include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino
> #else
> #include [...]

This fixed for me using LilyGo-T-Call-SIM800L.

ramante, 2020-07-29:
> There is an async version of this manager without the issues here (I'm already using it)
> https://github.com/alanswx/ESPAsyncWiFiManager

Hi Garito, thank you so much for the solution, works 100% to me. 
Regards,
Rogerio

Andrisetianabrata, 2021-10-30:
> A better solution imo is to rename the constant in WifiManager.h:
> 
> HTML_HEAD -> HTTP_HEAD_HTML
> 
> and then rename all the references in WifiManager.cpp:
> 
> HTML_HEAD -> HTTP_HEAD_HTML
> 
> I verified that this works. This way you are not breaking WebServer if it references that constant anywhere (or any other libs that depend on it)

Thanks mate thats work as well 👍👍❤️