zhouhan0126/WIFIMANAGER-ESP32 #19: esp32 arduino

SuryaTejMahankali, 2018-09-28:
this is the program the same autoconnect program
void setup() {
    Serial.begin(115200);

    WiFiManager wifiManager;
    wifiManager.autoConnect("AutoConnectAP");
    Serial.println("connected...yeey :)");
}

void loop() {
    // put your main code here, to run repeatedly:

}

Getting error:
_In file included from C:\Users\HP\AppData\Local\Temp\arduino_modified_sketch_722435\AutoConnect.ino:14:0:

C:\Users\HP\Documents\Arduino\libraries\WIFIMANAGER-ESP32-master/WiFiManager.h:36:22: error: 'const char HTTP_HEAD []' redeclared as different kind of symbol

 const char HTTP_HEAD[] PROGMEM            = "<!DOCTYPE html><html lang=\"en\"><head><meta name=\"viewport\" content=\"width=device-width, initial-scale=1, user-scalable=no\"/><title>{v}</title>";

                      ^

In file included from C:\Users\HP\Documents\Arduino\hardware\espressif\esp32\libraries\WebServer\src/WebServer.h:30:0,

                 from C:\Users\HP\Documents\Arduino\libraries\WIFIMANAGER-ESP32-master/WiFiManager.h:21,

                 from C:\Users\HP\AppData\Local\Temp\arduino_modified_sketch_722435\AutoConnect.ino:14:

C:\Users\HP\Documents\Arduino\hardware\espressif\esp32\libraries\WebServer\src/HTTP_Method.h:10:3: note: previous declaration 'HTTPMethod HTTP_HEAD'

   HTTP_HEAD    = 0b00100000,

   ^

Multiple libraries were found for "WiFiManager.h"
 Used: C:\Users\HP\Documents\Arduino\libraries\WIFIMANAGER-ESP32-master
 Not used: C:\Users\HP\Documents\Arduino\libraries\WiFiManager-0.14.0_

plz help struggling for almost 3 days.

bobermen1, 2018-10-03:
Copied from another reply

I was having this issue with the examples as well.
To fix it I replaced
```
#if defined(ESP8266)
#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino
#else
#include <WiFi.h>          //https://github.com/esp8266/Arduino
#endif

//needed for library
#include <DNSServer.h>
#if defined(ESP8266)
#include <ESP8266WebServer.h>
#else
#include <WebServer.h>
#endif
#include <WiFiManager.h>         //https://github.com/tzapu/WiFiManager
```

With
```
#include <WiFi.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>
```

standarddeviant, 2018-10-10:
I have the same issue. @bobermen1 which library/file did you edit?

shahidare, 2018-10-25:
Hi,

I did the changes it still seems not to be working.

I am getting the follow error message.

In file included from /Users/Sid/Documents/Arduino/libraries/WIFIMANAGER-ESP32-master/examples/AutoConnect/AutoConnect.ino:14:0:
/Users/Sid/Documents/Arduino/libraries/WIFIMANAGER-ESP32-master/WiFiManager.h:36:22: error: 'const char HTTP_HEAD []' redeclared as different kind of symbol
 const char HTTP_HEAD[] PROGMEM            = "<!DOCTYPE html><html lang=\"en\"><head><meta name=\"viewport\" content=\"width=device-width, initial-scale=1, user-scalable=no\"/><title>{v}</title>";
                      ^
In file included from /Users/Sid/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/libraries/WebServer/src/WebServer.h:30:0,
                 from /Users/Sid/Documents/Arduino/libraries/WIFIMANAGER-ESP32-master/examples/AutoConnect/AutoConnect.ino:12:
/Users/Sid/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/libraries/WebServer/src/HTTP_Method.h:10:3: note: previous declaration 'HTTPMethod HTTP_HEAD'
   HTTP_HEAD    = 0b00100000,
   ^
Multiple libraries were found for "WiFi.h"
 Used: /Users/Sid/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/libraries/WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
Multiple libraries were found for "DNSServer.h"
 Used: /Users/Sid/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/libraries/DNSServer
 Not used: /Users/Sid/Documents/Arduino/libraries/DNSServer---esp32-master
Multiple libraries were found for "WebServer.h"
 Used: /Users/Sid/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/libraries/WebServer
 Not used: /Users/Sid/Documents/Arduino/libraries/WebServer-esp32-master
exit status 1
Error compiling for board Node32s.