zhouhan0126/WIFIMANAGER-ESP32 #22: WIfiManager on ESP32 is not working

shahidare, 2018-10-25:
Hi,

I am trying to run the Autoconnect example from the wifimanager esp32.

I am getting the following error:

In file included from /Users/Sid/Documents/Arduino/libraries/WIFIMANAGER-ESP32-master/examples/AutoConnect/AutoConnect.ino:14:0:
/Users/Sid/Documents/Arduino/libraries/WIFIMANAGER-ESP32-master/WiFiManager.h:36:22: error: 'const char HTTP_HEAD []' redeclared as different kind of symbol
const char HTTP_HEAD[] PROGMEM = "<html lang="en"><meta name="viewport" content="width=device-width, initial-scale=1, user-scalable=no"/><title>{v}</title>";
^
In file included from /Users/Sid/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/libraries/WebServer/src/WebServer.h:30:0,
from /Users/Sid/Documents/Arduino/libraries/WIFIMANAGER-ESP32-master/examples/AutoConnect/AutoConnect.ino:12:
/Users/Sid/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/libraries/WebServer/src/HTTP_Method.h:10:3: note: previous declaration 'HTTPMethod HTTP_HEAD'
HTTP_HEAD = 0b00100000,
^
Multiple libraries were found for "WiFi.h"
Used: /Users/Sid/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/libraries/WiFi
Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
Multiple libraries were found for "DNSServer.h"
Used: /Users/Sid/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/libraries/DNSServer
Not used: /Users/Sid/Documents/Arduino/libraries/DNSServer---esp32-master
Multiple libraries were found for "WebServer.h"
Used: /Users/Sid/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/libraries/WebServer
Not used: /Users/Sid/Documents/Arduino/libraries/WebServer-esp32-master
exit status 1
Error compiling for board Node32s.

aphex008, 2018-10-31:
Same here on lolin d32

Compiling .pioenvs/lolin_d32/src/main.cpp.o
In file included from src/main.cpp:21:0:
lib/WIFIMANAGER-ESP32-master/WiFiManager.h:36:22: error: 'const char HTTP_HEAD []' redeclared as different kind of symbol
const char HTTP_HEAD[] PROGMEM            = "<!DOCTYPE html><html lang=\"en\"><head><meta name=\"viewport\" content=\"width=device-width, initial-scale=1, user-scalable=no\"/><title>{v}</title>";
^
In file included from /Users/darius/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/WebServer.h:30:0,
from src/main.cpp:19:
/Users/darius/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/HTTP_Method.h:10:3: note: previous declaration 'HTTPMethod HTTP_HEAD'
HTTP_HEAD    = 0b00100000,
^
Compiling .pioenvs/lolin_d32/libba5/WebServer/detail/mimetable.cpp.o
Compiling .pioenvs/lolin_d32/lib52d/WIFIMANAGER-ESP32-master/WiFiManager.cpp.o
In file included from lib/WIFIMANAGER-ESP32-master/WiFiManager.cpp:13:0:
lib/WIFIMANAGER-ESP32-master/WiFiManager.h:36:22: error: 'const char HTTP_HEAD []' redeclared as different kind of symbol
const char HTTP_HEAD[] PROGMEM            = "<!DOCTYPE html><html lang=\"en\"><head><meta name=\"viewport\" content=\"width=device-width, initial-scale=1, user-scalable=no\"/><title>{v}</title>";
^
In file included from /Users/darius/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/WebServer.h:30:0,
from lib/WIFIMANAGER-ESP32-master/WiFiManager.h:21,
from lib/WIFIMANAGER-ESP32-master/WiFiManager.cpp:13:
/Users/darius/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/HTTP_Method.h:10:3: note: previous declaration 'HTTPMethod HTTP_HEAD'
HTTP_HEAD    = 0b00100000,
^

lacneto, 2018-11-05:
I was having the same HTTP_HEAD problem and the solution was simpler than it looked. I just edited the WiFiManager.h files and WifiManager.cpp replacing (HTTP_HEAD) with (HTTP_HEADER) and there was no more compilation problem.

shahidare, 2018-11-05:
I did that but then wifi manager was not functioning for me.

lacneto, 2018-11-05:
Do you using libraries WebServer and DNSServer the same autor?
https://github.com/zhouhan0126/WebServer-esp32
https://github.com/zhouhan0126/DNSServer---esp32
Try it use this libraries together and see if working.
For me to be working perfectly.

Regards

shahidare, 2018-11-05:
For now I am using asyc version and it is working. ( other than wifi reset
)

I did tired the usual hacks but it doesn’t seem to be working. I am using
Mac for the arduino. I believe I have to try it with other machines as I
can see it has worked for many people.

Thank you for the help.

CraigH78, 2019-02-01:
@lacneto 
HTTP_HEAD to HTTP_HEADER worked for me also.  Thank you!

DaveCalaway, 2020-01-28:
In the Arduino IDE i must edited the file "WiFiManager.h" as @lacneto said. But i don't understand why this appends.

In case someone uses Platform IO, you just edit the platformio.ini adding:
```
lib_deps =
  https://github.com/zhouhan0126/WIFIMANAGER-ESP32.git
  https://github.com/zhouhan0126/DNSServer---esp32.git
  https://github.com/zhouhan0126/WebServer-esp32.git
```

sMr1dul, 2020-07-22:
just add __ to the HTTP_HEAD[] variable in WifiManager.h and replace HTTP_HEAD  to __HTTP_HEAD in WiFiManager.cpp. in arduino WifiManager library.
[image]
[image]

netphantm, 2020-09-26:
> I was having the same HTTP_HEAD problem and the solution was simpler than it looked. I just edited the WiFiManager.h files and WifiManager.cpp replacing (HTTP_HEAD) with (HTTP_HEADER) and there was no more compilation problem.

Had to do this, also on an ESP8266, works now.

MorhafJAela, 2020-09-27:
> I was having the same HTTP_HEAD problem and the solution was simpler than it looked. I just edited the WiFiManager.h files and WifiManager.cpp replacing (HTTP_HEAD) with (HTTP_HEADER) and there was no more compilation problem.

It is very useful!
thanks

scarrrr316, 2021-04-15:
It is very useful!

viniandrs, 2021-07-15:
> I was having the same HTTP_HEAD problem and the solution was simpler than it looked. I just edited the WiFiManager.h files and WifiManager.cpp replacing (HTTP_HEAD) with (HTTP_HEADER) and there was no more compilation problem.

After I did this it worked for me :)