zhouhan0126/WIFIMANAGER-ESP32 #26: How to check to which network you are connected ?

GeorgeFlorian, 2019-02-01:
My ESP32 connects to a WiFi Network and I have no idea which one. How can I see this ? I've read the documentation and looked inside the wifiManager.h and .cpp files and couldn't find a function to do that.

I've never seen the web configuration portal. 

Also, how can I modify the web configuration portal using SPIFFS file?

Thank you !

GeorgeFlorian, 2019-02-21:
I've seen little to no activity here. Is the author not offering any support for this library ?

LukasAV, 2019-04-21:
You can display the SSID of the current connected network with `wifiMan.getSSID()` where wifiMan is your WiFiManager object.
To prevent connecting to the network use `startConfigPortal("APName");` to force the ESP to open up an access point and captive portal to connect to another network.

GeorgeFlorian, 2019-04-22:
> You can display the SSID of the current connected network with `wifiMan.getSSID()` where wifiMan is your WiFiManager object.
> To prevent connecting to the network use `startConfigPortal("APName");` to force the ESP to open up an access point and captive portal to connect to another network.

Don't worry ! I never used this library ! You shouldn't use an unsupported library !

closed 2019-04-22 by GeorgeFlorian

LukasAV, 2019-04-22:
Is there an alternative WiFiManager library with support for the ESP32?

GeorgeFlorian, 2019-04-26:
> Is there an alternative WiFiManager library with support for the ESP32?

I really do not know ! Haven't looked for one. I use `WiFi.h` that comes standard with arduino-core for ESP32.

shariq-azim, 2021-04-09:
> You can display the SSID of the current connected network with `wifiMan.getSSID()` where wifiMan is your WiFiManager object.
> To prevent connecting to the network use `startConfigPortal("APName");` to force the ESP to open up an access point and captive portal to connect to another network.

Hi there,
I was trying to print the ssid of the network last connected to/or set in wifi manager. but it doesn't compiles

eg.. WiFiManager wifiManager;
  wifiManager.setTimeout(60);
  
 wifiManager.setDebugOutput(true);
  

  if(!wifiManager.autoConnect("AutoConnectA0P")) {
    Serial.println("failed to connect and hit timeout");
    delay(3000);
} 
   Serial.println("start");
  Serial.println( wifiManager.getSSID());
   // Serial.println( WiFiManager.getConfigPortalSSID() );

ssid= wifiManager.getSSID();

i get error in last line  stating "'class WiFiManager' has no member named 'getSSID'"