zhouhan0126/WIFIMANAGER-ESP32 #28: wifi scan not showing anything

Johan-de-Wit, 2019-02-07:
when the config portal starts it does not show any available WiFi stations. Why not?

jroblesruiz, 2019-02-25:
Im having the same problem

gregoryalex, 2019-03-22:
I'm having the exact same problem...

wdouglas1999, 2019-04-05:
Hi all, I’m also having the same problem. 
Any suggestions yet on what could be the issue?

Phoenix260, 2019-04-08:
I added: 

```
    // Set WiFi to station mode and disconnect from an AP if it was previously connected
    WiFi.mode(WIFI_AP_STA);
    WiFi.disconnect();
    delay(100);
    Serial.println("WiFi to station mode and disconnect from an AP if it was previously connected");

    Serial.println("scan start");
    // WiFi.scanNetworks will return the number of networks found
    int n = WiFi.scanNetworks();
    Serial.println("scan done");
    if (n == 0) {
        Serial.println("no networks found");
    } else {
        Serial.print(n);
        Serial.println(" networks found");
        for (int i = 0; i < n; ++i) {
            // Print SSID and RSSI for each network found
            Serial.print(i + 1);
            Serial.print(": ");
            Serial.print(WiFi.SSID(i));
            Serial.print(" (");
            Serial.print(WiFi.RSSI(i));
            Serial.print(")");
            Serial.println((WiFi.encryptionType(i) == WIFI_AUTH_OPEN)?" ":"*");
            delay(10);
        }
    }
    Serial.println("");
```

before the wifiManager.startConfigPortal() or the wifiManager.autoConnect() 

I think the first 2 lines is really what fixes it ... but the rest gives you some troubleshooting to go by. My issue was that the number of networks returned as -2

This should be added to the void WiFiManager::handleWifi(boolean scan) right after the 
int n = WiFi.scanNetworks();
if(n == -2) DEBUG_WM(F("scan not triggered"));
if(n == -1) DEBUG_WM(F("scan not finished"));

-2 is scan is not triggered and -1 is that it didn't finish