zhouhan0126/WIFIMANAGER-ESP32 #29: curent lib is not compiled

planer-pro, 2019-02-10:
i got strange error while compiling: 

WiFiManager.h: 36:22: error: 'const char HTTP_HEAD []' redeclared as different kind of symbol const char HTTP_HEAD[] PROGMEM = "<!DOCTYPE html><html lang=\"en\"><head><meta name=\"viewport\" content=\"width=device-width, initial-scale=1, user-scalable=no\"\><title>{v}<\title>"

how to fix it who know?

kc2uez, 2019-02-17:
I had the same problem. I edited the WiFiManager.h and WiFiManager.c files replacing HTTP_HEAD with HTTP_HEAD_ESP32. Be aware there is a HTTP_HEAD_END too, so be careful. It needs to be changed on like 6 places or so.

davefiddes, 2019-11-03:
There looks to be a good fix for this in PR #24. Would be good to get this merged.