zhouhan0126/WIFIMANAGER-ESP32 #40: Error compiling for board ESP32 Dev Module.

hiva9816, 2020-04-02:
when i want to compile one of [WIFIManager](https://github.com/zhouhan0126/WIFIMANAGER-ESP32) library named AutoConnect , have faced to the following error. 

exit status 1
Error compiling for board ESP32 Dev Module.

i should mention that this error shown by this code. and other codes with esp32 are compiling completely  
can any one help me by this error?

lloydrichards, 2020-04-15:
Have you made sure that the ESP32 is correctly setup on your device? ie run a blink sketch to make sure the ports and connection are working.

Next, have you installed the correct libraries for this? the WebServer and DNSserver are not the one found in the arduino library manager and must be installed manually from Zhouhan's repo.

Next, can you compile the sketch without uploading it? Are there any errors in the code or is it just when you go to upload it to the ESP32

jaketeater, 2020-04-24:
I have a similar issue -

When compiling (Verify Button in the Arduino IDE) this example: https://raw.githubusercontent.com/zhouhan0126/WIFIMANAGER-ESP32/master/examples/AutoConnect/AutoConnect.ino for the Board 'ESP 32 Dev Module', I get the error below.

(I have downloaded both the DNSServer---esp32 and WebServer-esp32 repos and put them in my Library directory)

I assume I've made a simple mistake and just don't know enough to recognize it. Any help would be appreciated.

```
Arduino: 1.8.11 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

In file included from C:\Users\thete\Repos\raspi-radio\esp\firmware\firmware.ino:12:0:

C:\Users\thete\Documents\Arduino\libraries\WebServer-esp32-master\src/WebServer.h:206:15: error: cannot declare field 'WebServer::_server' to be of abstract type 'WiFiServer'

   WiFiServer  _server;

               ^

In file included from C:\Program Files (x86)\Arduino\libraries\WiFi\src/WiFi.h:32:0,

                 from C:\Users\thete\Repos\raspi-radio\esp\firmware\firmware.ino:4:

C:\Program Files (x86)\Arduino\libraries\WiFi\src/WiFiServer.h:31:7: note:   because the following virtual functions are pure within 'WiFiServer':

 class WiFiServer : public Server {

       ^

In file included from C:\Users\thete\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/Arduino.h:152:0,

                 from sketch\firmware.ino.cpp:1:

C:\Users\thete\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/Server.h:28:18: note: 	virtual void Server::begin(uint16_t)

     virtual void begin(uint16_t port=0) =0;

                  ^

exit status 1
Error compiling for board ESP32 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

ZebNZ, 2020-07-10:
@jaketeater did you fix the problem?

If not, ensure that you declared the libraries at the top of your .ino file like this (for ESP32 only). 
#include <FS.h>
#include "SPIFFS.h"
#include <WebServer.h>
#include <DNSServer.h>
#include <WiFi.h>
#include <WiFiManager.h">
#include <ArduinoJson.h>

AjwadWaie25, 2022-04-29:
I have an issue to compile my code, at first the problem is because multiple library detected for wifi.h. Then I deleted one of the library and I still having the same problem

Arduino: 1.8.19 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None

In file included from C:\Users\amatw\Documents\Arduino\libraries\Blynk\src/Blynk/BlynkApi.h:37:0,

                 from C:\Users\amatw\Documents\Arduino\libraries\Blynk\src/BlynkApiArduino.h:14,

                 from C:\Users\amatw\Documents\Arduino\libraries\Blynk\src/BlynkSimpleEsp32.h:20,

                 from C:\Users\amatw\Desktop\Coding\Testing_ADS1115_with_Sensor_Soil\Testing_ADS1115_with_Sensor_Soil.ino:3:

C:\Users\amatw\Documents\Arduino\libraries\Blynk\src/Blynk/BlynkTimer.h:36:21: error: redefinition of 'class BlynkTimer'

 #define SimpleTimer BlynkTimer

                     ^

C:\Users\amatw\Documents\Arduino\libraries\SimpleTimer/SimpleTimer.h:10:7: note: in expansion of macro 'SimpleTimer'

 class SimpleTimer {

       ^

C:\Users\amatw\Documents\Arduino\libraries\Blynk\src/Blynk/BlynkTimer.h:36:21: error: previous definition of 'class BlynkTimer'

 #define SimpleTimer BlynkTimer

                     ^

C:\Users\amatw\Documents\Arduino\libraries\Blynk\src/Blynk/BlynkTimer.h:41:7: note: in expansion of macro 'SimpleTimer'

 class SimpleTimer {

       ^

exit status 1

Error compiling for board ESP32 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.